quartiq/stabilizer #564: Relay aka. MCP23008 I2C shift register driver design

nkrackow, 2022-07-22:
The Driver outputs will feature a relay arrangement to short the laser diode at startup and for some additional protection mechanisms.
These relays are controlled by a I2C shift register/IO expander on Driver. 

I could only find this ["capsule" crate for the MPC23008](https://docs.tockos.org/capsules/mcp230xx/index.html) which has to be used with the [Tock](https://github.com/tock/tock) afaict. 

Since we currently only use a small fraction of the MPC23008 functionality (just toggeling outputs), I would lean towards writing a minimal combined Relay/MPC23008 driver that exposes functions like `relays.ground_output()` or `relays.connect_output()`. 
The driver also has to check for the relay logic (sometimes one cannot open if the other is in a certain state etc.). I would just have two booleans that track the physical relay state and the the driver functions checking and returning errors if the desired function cannot be performed.

nkrackow, 2022-07-22:
Oh I just saw that Stabilizer already uses the [MCP23017 driver](https://crates.io/crates/mcp23017)! 
At first sight it looks like this chip has two times the IO banks and therefore we should be able to simply use it and just never use the second IO bank.

jordens, 2022-07-25:
Agreed. Either add support for MCP23008 (through banked mode) to `mcp23017` via PR or make your own `mcp23008` crate.

nkrackow, 2022-08-01:
About the actual relay logic: My plan is to have the following 3 state state machine.
[image]
I would then use the smlang crate for the actual implementation, even though I'm just starting to get what the benefits of using this crate would be here. Or I could just use a `set_state()` function that has a big match statement with current and next state or just the have transitions as functions and then check if the current state allows the transition.
Are the names for states and transitions reasonable? I'm happy to change them but I couldn't think of anything better right now..

nkrackow, 2022-08-02:
Actually, do we really need `ConnectedGrounded` as a state? Aka the driver output stage connected to the laser diode but also still grounded? Because otherwise we could just have the two states `Shunted` and `Connected` for each Driver output and pass through `ConnectedGrounded` during the transitions.

nkrackow, 2022-08-02:
New plan: State machine with 4 states that get traversed linearly in both directions. Leftmost and rightmost states are permanent until exited by `enable(), disable()`, while the intermediate states are temporary and exit timing is provided by fixed rtic scheduling delays to wait for slow relays.
[image]

nkrackow, 2022-08-09:
The design of the relay driver got merged in https://github.com/quartiq/stabilizer/pull/582.
The state machine now has separate states for the enabling/disabling transitions.

closed 2022-08-09 by nkrackow