quartiq/stabilizer #585: CPU temperature measurement unstable

ryan-summers, 2022-08-03:
The CPU-reported temperature measurements do not appear to be very stable:
[image]

It was originally assumed that the ADC clock being out of spec as specified in https://github.com/quartiq/stabilizer/issues/571 was the cause, buy even after fixes in #584, this issue does not appear to be resolved.

jordens, 2022-08-03:
Let's try an example from the Hal and see how that behaves.

nkrackow, 2022-08-03:
> Let's try an example from the Hal and see how that behaves.

I can do that tomorrow.

ryan-summers, 2022-08-03:
Interestingly enough, while it varies significantly, it does appear to measure around an average point:
[image]

Earlier today, it seemed initially stable around ~40 C and slowly increased to ~55 while powered.

jordens, 2022-08-03:
Ah. That looks similarly noisy to Ryan's data above.
This is the https://github.com/stm32-rs/stm32h7xx-hal/blob/master/examples/temperature.rs example, right?

nkrackow, 2022-08-04:
> Ah. That looks similarly noisy to Ryan's data above. This is the https://github.com/stm32-rs/stm32h7xx-hal/blob/master/examples/temperature.rs example, right?

Natively this example gives a rather unrealistic 84 °C average, with ~ 3-4 °C spread. I'll dig a bit deeper.

jordens, 2022-08-04:
Make sure you supply the correct reference voltage (and set it up for external reference).

nkrackow, 2022-08-04:
I think the ADC can only be used with the external reference. Changing that to the correct value yields temperatures in the range of 32-34 °C. I can bring that up to ~50 °C with my flatmates hairdryer.
So this seems like sensible data to me. I'll start looking into differences to the way the stabilizer code sets things up.

ryan-summers, 2022-08-04:
> Ah. That looks similarly noisy to Ryan's data above. This is the https://github.com/stm32-rs/stm32h7xx-hal/blob/master/examples/temperature.rs example, right?

No, sorry, I should have been clear. That was stabilizer telemetry after I left the device running on my desk all day.

jordens, 2022-08-04:
> So this seems like sensible data to me. I'll start looking into differences to the way the stabilizer code sets things up.

What are you looking for? If the result is the same, I wouldn't expect to find a difference.

> No, sorry, I should have been clear. That was stabilizer telemetry after I left the device running on my desk all day.

Ah. I thought that post was from Norman. Sure It'll average down. If it didn't that would be a different problem.

jordens, 2022-08-04:
> I think the ADC can only be used with the external reference

Do check the RM for VREFBUF. But looks like unless you touch it it'll use the external reference.

nkrackow, 2022-08-04:
Hmm the more I mess around with this, the more I think this could be the actual temperature of the chip at the point of the sensor fluctuating.
If I just print the temperature in an infinite loop again after the adc setup in the stabilizer setup, I see almost the same temperatures with the same fluctuation (~2-3 °C) as I get with the example. However, this is only the case if I set the clock speed to the same 400 MHz in both! If I leave the example at 100 MHz, it's about 4 °C lower. The stabilizer code already enabled some more clocks and peripherals so the remaining discrepancy of about 1 °C is plausible.
The CPU temperature is normally reported during the whole complex program runs and displays a larger spread of values. So maybe that's due to more dynamic things going on on the chip? Also my deviation is somehow a bit less than Ryans.
[image]

jordens, 2022-08-04:
@nkrackow does the temperature in the example fluctuate?

nkrackow, 2022-08-04:
> @nkrackow how exactly does the temperature in the example fluctuate? 
This is how the raw data looks like. Roughly 10 Hz sample rate. I don't know where the fluctuations come from.
[image]

jordens, 2022-08-04:
That's 0.5 K rms. But this is still a bug (silicon or hal). 3 K pk-pk fluctuation in 100 ms on an otherwise steady-state chip is just not plausible.
In the example this is with 4 MHz pll2_p and thus 12.5 times slower ADC than for stabilizer with 50 MHz. But the sampling time is 32 (if I'm not mistake, that still is a bug and violates the RM) and we do 810 cycles, thus absolute sampling is shorter. Could you try whether T810 makes a difference in the example?
Otherwise I'd be inclined to closing this as either a HAL or silicon issue.

nkrackow, 2022-08-04:
> Could you try whether T810 makes a difference in the example?
I did, it doesn't. Only when you make it very short it does.

jordens, 2022-08-09:
Let's close this then.

closed 2022-08-09 by jordens