quartiq/stabilizer #607: Stream: UDP not supported by ProactorEventLoop on Windows

pmldrmota, 2022-08-28:
The default `asyncio` event loop on Windows doesn't support UDP (see https://docs.python.org/3.6/library/asyncio-eventloops.html#windows).

To make the following code work on Windows,
https://github.com/quartiq/stabilizer/blob/831a9ba747b51241936b13a2df23753c03cc9dd5/py/stabilizer/stream.py#L96-L97

one has to change the event loop. This can be achieved using
```python
new_loop = asyncio.SelectorEventLoop()
asyncio.set_event_loop(new_loop)
```

Note that, with the `ProactorEventLoop`, the stream still starts and processes data, but at shutdown, the task hangs and never completes.

dnadlinger, 2022-08-28:
It's supported also using the proactor event loop on Python 3.8+ (released almost three years ago).

ryan-summers, 2022-08-29:
Even so, we should explicitly state this version dependency in docs. I don't see it mentioned anywhere in https://quartiq.de/stabilizer/usage.html yet - it seems prudent to call this out as a minimum required version (and test it in CI)

jordens, 2022-08-29:
@pmldrmota do you want to prepare a PR that requires python >= 3.8?

ryan-summers, 2022-08-29:
Fixed in https://github.com/quartiq/stabilizer/pull/608. Closing.

closed 2022-08-29 by ryan-summers