quartiq/stabilizer #647: Peripheral clock seems to be off.

nkrackow, 2022-11-11:
While migrating to the HAL V13.1 I expected the program to panic because the HAL now [checks that the internal ADC core clock is not above 80 MHz in out case](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/adc.rs#L578-L583).
[Point in the datasheet](https://www.st.com/resource/en/reference_manual/dm00314099-stm32h742-stm32h743-753-and-stm32h750-value-line-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf#page=353).

https://github.com/quartiq/stabilizer/blob/3562cdc377093fbfabdef91c47467ef83d024654/src/hardware/setup.rs#L278
Using a debug print in the HAL adc setup code I determined that the ADC core and therefore `per_ck` is clocked at 64 MHz and not at 100 MHz as the line above suggests.
Is this plausible with other peripherals or am I missing something?

jordens, 2022-11-11:
Maybe its just integer divider and rounding.

nkrackow, 2022-11-14:
Ok, so not a problem? I thought because it's some `TIMER_FREQUENCY` it might matter that it's actually that value.

jordens, 2022-11-14:
Not sure. The timer tick is (also) set at 
https://github.com/quartiq/stabilizer/blob/3562cdc377093fbfabdef91c47467ef83d024654/src/hardware/setup.rs#L324
and other places. I don't know off hand what clock the timers are clocked by.

Should definitely be checked and either fixed or commented.
That said I was also not entirely convinced I understood the internal ADC clocking frequencies and constraints.

closed 2022-11-22 by bors[bot]