quartiq/stabilizer #654: Errer print leads to spurious panics

nkrackow, 2022-11-28:
https://github.com/quartiq/stabilizer/blob/e8116b34a254073bbfb0e7b163989f34af5504e6/src/hardware/driver/mod.rs#L107-L110

ryan-summers, 2022-12-06:
I'm somewhat confused why this would be happening. We take explicit steps to ensure that ISRs are not disabled during log printing/formatting.

The only reason this could be causing a panic would be if the `laser_interlock` (or something outside of it) has elevated the ceiling ISR priority to that equivalent of the DMA ISR. What's the ISR ceiling in the middle of that call? Is something elevating the priority ceiling somehow that isn't clear?

I can't see anything immediately in main, but that's what I would expect to be the root cause. Formatting can take some time, but the DMA process _should_ be able to interrupt the formatting. Something seems like it's not allowing the DMA to interrupt

ryan-summers, 2022-12-07:
We explicitly update logging to not use critical sections: https://github.com/quartiq/stabilizer/blob/master/src/hardware/setup.rs#L228

nkrackow, 2023-02-17:
I've revisited this now and I understand your point @ryan-summers. I sometimes get an ADC FIFO overflow when I uncomment the error print and trigger an interrupt condition. This should not happen since the `process` task with the ADC handling is higher priority and should interrupt the logging.

nkrackow, 2023-02-17:
Anyways, since Driver development is further along now and the mqtt interlock reporting looks good with the same info as the error log I think we don't need it any more and I'll close this.

closed 2023-02-17 by nkrackow