quartiq/stabilizer #674: reconnect to flapping broker

jordens, 2023-02-06:
We had a case where the MQTT broker (IP and port) was flapping and stabilizer did not reconnect. The sequence was:

* broker reachable, then
* ip not reachable for 2 minutes, then
* ip reachable but no broker running for 1 minute, then
* broker reachable for 20 seconds,
* then ip unreachable for 1 minute,
* then broker reachable

Stabilizer did not reconnect within that time and within 5 minutes after that. DHCP was fine during during that sequence.
Restarted stabilizer and it reconnected.
Will try to isolated and reproduce and triage.

jordens, 2023-02-06:
In a quick test it was enough to restart the broker to trigger the bug. I.e. it didn't reconnect.

ryan-summers, 2023-05-09:
Below is a screenshot of the captured traffic as the broker disconnects:
[image]

It's worth noting that I wasn't able to reproduce this with a container-based broker, but when running the broker as a service on a Raspberry Pi, it was reproducible by stopping/starting the mosquitto service.

ryan-summers, 2023-05-09:
When running via docker, where this issue does not present itself, we see different socket behavior on Wireshark:
[image]

### Analysis

It looks like in both cases, when the broker is shutting down, it sends a `FIN. ACK` to the device, which causes the socket to be closed. This is confirmed on both images. Stabilizer responds with an `ACK` to the `FIN` and appears to close the sockets (the socket number is not reused). Immediately, it appears that Stabilizer opens new sockets (new socket numbers) and attempts to connect to the broker with an `SYN` packet.

In the docker instance, the PC responds with a `SYN, ACK`, and the TCP connection establishes long enough for Stabilizer to send an MQTT Connect message. However, the PC then sends renewed `FIN, ACK` sequences, repeatedly aborting the established TCP connections. This occurrs repeatedly until the broker is restarted.

#### Service-Based Analysis

The above doesn't appear to be the case for the RPi scenario. In this case, Stabilizer detects the initial shutdown and closes sockets. It then immediately attempts to reconnect, but gets a `RST, ACK` in response to its initial SYN.

ryan-summers, 2023-05-09:
This is caused by a bug in the minimq TCP state machine. If the socket loses connection before the `Connect` message can be sent, it will be perpetually stuck in the `Restart` state, but will not be ableto send the `Connect` packet.

PR #731 (2023-06-22): Fixing MQTT reconnection issues, updating smoltcp version
This PR:
* Updates Smoltcp to 0.9
* Fixes #674 

Prerequisites:
- [x] https://github.com/quartiq/smoltcp-nal/pull/42 merged and released
- [x] https://github.com/quartiq/miniconf/pull/145 merged and released

I also incidently had to bump the HAL to the `master` branch because smoltcp 0.9 support isn't released yet.
closed 2023-06-22 by ryan-summers