quartiq/stabilizer #746: Stabilizer + Pounder panics without debug probe connection

ryan-summers, 2023-07-06:
If the Pounder+Stabilizer combination is flashed and rebooted, it has been reported that it will panic without the debugger attached (as indicated by the red LEDs powered)

Upon using a debugger reset, the panic is resolved.

First reported by @jordens

ryan-summers, 2023-07-07:
The panic appears to be occurring because the AD9959 register CSR readback is mismatching the configured value. This reminds me of https://github.com/quartiq/stabilizer/issues/101, and I'll continue investigating what's happening here.

jordens, 2023-07-07:
Pretty sure the SPI interface works fine without the refclk input as it's a bootstrap requirement for e.g. xtal setup. Hence I don't see the requirement. Also can't find it in the datasheet.
The only thing is the master reset vs sysclk requirement. Maybe just moving the reset after the identification is enough. But that's fragile as the register we use isn't just some identification register.
If that doesn't help, at boot we always use the internal oscillator. Just waiting for it to come up should do it then. I.e. increase the reset delay to 1 ms.

ryan-summers, 2023-07-07:
The IO_Update and MASTER_RESET signals require a minimum 1 sync clock period:
[image]

Without I/O update, the register configurations do not become active:
[image]

Thus, if we have no SYNC-CLK (which is derived from REF_CLK), there is no way to modify internal registers.

ryan-summers, 2023-07-07:
Additionally, the clock selection is not done over the SPI interface, it's done via an external `CLK_MODE_SEL` pin:
[image]

Thus, if we have this driven low permanently, we always use an external oscillator, which may not be fully on.

jordens, 2023-07-07:
Clock configuration is done through FR1. If that's not configured it's somewhat unclear that a useful sysclk is available, ergo the SPI interface should work without clock. My guess is that we're actually running into the reset pulse timing issue.
CLK_MODE_SEL isn't relevant here as there is no safe way to have that high. There is no other oscillator.

ryan-summers, 2023-07-07:
The `CLK_MODE_SEL` is used to select between the internal oscillator of the AD9959 and the external REFCLK inputs:
[image]

If we have CLK_MODE_SEL driven permanently low, the on-chip oscillator is disabled and the AD9959 is clocked entirely through REFCLK inputs (which are not oscillating)

FR1 does not control selection between the internal clock and the external clock, it only allows powering down the REFCLK input pins (but they are powered up by default)

ryan-summers, 2023-07-07:
Ah you're correct here. However, the point stands that there is no clock input on REF-CLK. And the data sheet explicitly says that there has to be a single SYNC-CLK cycle to register the I/O_Update signal (because it is synchronous). Thus, without input on REF-CLK, there is no SYNC-CLK, and the I/O_Update signal will never be registered.

FR1 configuration is not required to generate a SYNC-CLK source - there just has to be _any_ type of oscillation on REFCLK (either via crystal or oscillator, as selected by CLK_MODE_SEL)

jordens, 2023-07-07:
FR1 is definitely required! It's not like anything always generates a valid sysclk of sync clk. Imagine having it configured to use the PLL. Then the refclock frequency changes which can make FR1 invalid such that the PLL doesn't lock or isn't stable. Then you want to change FR1 but you don't have a valid clock.
The only hope would be that master reset is robust and works well enough and in absolutely every situation to set the clocking circuitry to pass-through to do the rest of its reset machinery (which then again needs a clock).

ryan-summers, 2023-07-07:
> FR1 is definitely required! It's not like anything always generates a valid sysclk of sync clk. Imagine having it configured to use the PLL. Then the refclock frequency changes which can make FR1 [...]

While my assessment of your statements agrees that there could be problems if you change your refclk, there's nothing in the datasheet (or observed behavior) that indicates that this is the case. Attempting to access and modify device registers while the REFCLK inputs are idle does not appear to result in registers sticking. I can try larger I/O_Update periods, but I doubt that would resolve the issue (since there's no clock at all)

jordens, 2023-07-07:
We should definitely just make sure we have (a) the oscillator up, (b) selected and (c) clocking before the reset. The rest shouldn't need changes AFAICS.

PR #748 (2023-07-07): Fixing pounder bootup issues
Fixes #746 by waiting a fixed delay before initializing the AD9959 to allow the Pounder clock to stabilize
closed 2023-07-07 by jordens