quartiq/stabilizer #786: stream to multicast

jordens, 2023-09-25:
* investigate using multicast for streaming
* solves the ARP cache lifetime issue and related potential dropouts
* allows simultaneous reception/logging
* probably no need to join any multicast group as we only send (before #613 )

ryan-summers, 2023-10-11:
@jordens I took a look, and the main question here is to figure out what broadcast address to use.

Some options:
1. Leave the current IP-configuration as-is, such that the client just requests a broadcast address (requires no firmware changes)
2. Fix the firmware to broadcast to some fixed UDP multicast address that we select and update tools to point at that

It seems like (1) may be more supportive of multiple Stabilizers on the same network. I also suspect this would be relatively trivial to update the test tools to do this.

jordens, 2023-10-11:
(1) is the way to go.

PR #788 (2023-10-12): Adding support for multicast stream data
This PR fixes #786 by updating the python tools to direct Stabilizer to stream to multicast UDP addresses. Note that there is no firmware change required for this.

Local run of the streaming HITL:
```
$ python311 hitl/streaming.py --broker 10.35.20.1 dt/sinara/dual-iir/04-91-62-d2-a8-6f
INFO:gmqtt.mqtt.protocol:[CONNECTION MADE]
INFO:gmqtt.mqtt.package:[SEND SUB] 1 [b'dt/sinara/dual-iir/04-91-62-d2-a8-6f/response']
INFO:gmqtt.mqtt.package:[SEND SUB] 2 [b'dt/sinara/dual-iir/04-91-62-d2-a8-6f/settings/#']
INFO:gmqtt.mqtt.handler:[SUBACK] 1 (0,)
INFO:mini [... cut]
closed 2023-10-12 by jordens