quartiq/stabilizer #804: Stabilizer + Pounder (still) panics on bootup

ayushagrawal2, 2023-11-10:
Stabilizer v1.3.3 panics when booted with Pounder v1.2 connected. The panic is caused upon setting the gpio pins in `hardware/pounder/mod.rs:388`, which returns a `NotAcknowledge` error before it is mapped to an I2c error. 

I thought it might be similar to #746 despite the fix in #748, but further increasing the delay to 100ms did not help.

On running `cargo run --release --bin dual-iir`, I get

```
    Finished release [optimized + debuginfo] target(s) in 0.14s
     Running `probe-run --chip STM32H743ZITx --speed 30000 target/thumbv7em-none-eabihf/release/dual-iir`
(HOST) INFO  flashing program (265 pages / 265.00 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
INFO - Starting
INFO - EUI48: 80-34-28-5e-f3-5e
INFO - Found Pounder
panicked at src/hardware/setup.rs:857:10:
called `Result::unwrap()` on an `Err` value: I2c
────────────────────────────────────────────────────────────────────────────────
(HOST) WARN  no RAM region appears to contain the stack; probe-run can't determine if this was a stack overflow
stack backtrace:
   0: HardFaultTrampoline
      <exception entry>
   1: cortex_m::asm::inline::__udf
        at /home/ion/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/../asm/inline.rs:181:5
   2: cortex_m::asm::udf
        at /home/ion/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/call_asm.rs:11:43
   3: rust_begin_unwind
        at src/hardware/mod.rs:113:5
   4: core::panicking::panic_fmt
        at /rustc/189d6c71f3bb6c52113b5639a80839791974fd22/library/core/src/panicking.rs:72:14
   5: core::result::unwrap_failed
        at /rustc/189d6c71f3bb6c52113b5639a80839791974fd22/library/core/src/result.rs:1653:5
   6: stabilizer::hardware::setup::setup
   7: dual_iir::app::init
        at src/bin/dual-iir.rs:211:14
   8: dual_iir::app::rtic_ext::main::{{closure}}
        at src/bin/dual-iir.rs:177:1
   9: dual_iir::app::rtic_ext::main::__rtic_init_resources
        at src/bin/dual-iir.rs:177:1
  10: main
  11: Reset
(HOST) WARN  call stack was corrupted; unwinding could not be completed
(HOST) ERROR the program panicked
```
The stack trace is identical with the `-- --no-flash` option after a panic.

ryan-summers, 2023-11-10:
Interesting. This would indicate an I2C communication error with the MCP23017 GPIO expander. But that hasn't changed based on my knowledge between the versions. Does this happen every time you run firmware, or just after power cycling the device?

ryan-summers, 2023-11-10:
Ah, this appears to be happening because pounder v1.2 incorporates a GPIO expander swap that we haven't yet implemented firmware for:
[image]

As such, we just don't support pounder v1.2 yet. I can take a look at adding support later.

ayushagrawal2, 2023-11-10:
Ah great, that's useful to know. Thank you!

dnadlinger, 2023-11-10:
Huh, but from https://github.com/sinara-hw/Pounder/issues/110 I would assume that you guys have been running 1.2 internally – has this just not made it onto a public branch yet?

ryan-summers, 2023-11-10:
It's possible, it's also possible that @jordens has a variant with the MCP23017 installed - I'm not sure.

PR #837 (2024-01-16): rj/pounder v1.2
- pounder v1.2 dds reset
- pounder v1.2: pca9359
- use own pca9539 due to alloc
- use crate
- refactor IoExpander
- refactor i2c setup on pounder
- pounder: style

close: #804
closed 2024-01-16 by jordens