quartiq/stabilizer #817: USB device not detected on Windows

juhaszp95, 2023-11-20:
Hello,

I've just tried setting up a stabiliser using the docs just released, configuring the MQTT broker using pyserial running on a Windows PC. I've managed to flash the firmware on the device alright.

However, when I plug the USB cable in to set the MQTT broker, Windows gives these error messages:
[image]
[image]

Also, when I launch pyserial, it doesn't really show up in the COM ports (as far as I can tell).

I wonder if there's some driver to install or similar to get this to work? The documentation doesn't say much about this.

jordens, 2023-11-20:
Potentially https://github.com/rust-embedded-community/usb-device/pull/129
It'll need trickling down the dependency chain.
Alternatively set the broker per compilation option as in previous releases.

ryan-summers, 2023-11-20:
This is indeed the above issue. We're working on getting it patched up through the dependency chain and will likely take a few days.

jordens, 2023-11-21:
Try again with #813

juhaszp95, 2023-11-21:
I've just cloned the repo again from scratch, compiled and flashed the firmware, but unfortunately I'm still getting the same errors.

When I run pyserial, I just get this:
[image]

It's the same whichever port I select - it just hangs there and doesn't do anything.

ryan-summers, 2023-11-21:
Yeah, nothing will show up by default, you have to press `Enter` to get the menu currently. I should mention this in the docs.

juhaszp95, 2023-11-21:
I'm afraid nothing happens when I press Enter (whichever port I choose).

I still get these errors mentioned earlier regarding Windows saying 'USB device not recognised', so I'm not even sure either of these ports is the right one.

juhaszp95, 2023-11-21:
> Potentially [rust-embedded-community/usb-device#129](https://github.com/rust-embedded-community/usb-device/pull/129) It'll need trickling down the dependency chain. Alternatively set the broker per compilation option as in previous releases.

Also, I've just tried doing this again, but this somehow also doesn't work for me any longer (it worked with previous versions) - I need to hardcode the MQTT broker address in `flash.rs`.

juhaszp95, 2023-11-29:
Hello - I wonder if you were able to have a look at this?

ryan-summers, 2023-11-29:
Unfortunately no. It works fine on my machine and in the past, I have required detailed USB logs to understand why the USB enumeration fails in the past.

I can look into getting that implemented to help here, but this is not a trivial issue to solve currently

ryan-summers, 2023-11-29:
@juhaszp95 Can you please provide all of the steps that you are using to build and then flash the device? I.e. provide all the command line steps that you run _exactly_. That may enlighten us to a potential issue.

juhaszp95, 2023-12-01:
The commands I run mirror exactly what is written at https://quartiq.de/stabilizer/setup.html.

At the beginning, the device is powered off and disconnected from the PC. The commands I run, in this order:
- `rustup update`
- `rustup target add thumbv7em-none-eabihf`
- `cargo install cargo-binutils`
- `rustup component add llvm-tools-preview`

These commands run without any error, saying that these components have already been installed (given I installed them previously when I flashed the stabiliser on previous occasions).

I then run:
- `git clone https://github.com/quartiq/stabilizer`
- `cd stabilizer`
- `cargo build --release`
- `cargo objcopy --release --bin dual-iir -- -O binary dual-iir.bin`

These commands finish OK. I then connect the stabiliser using an ST-Link probe and power it up. I then run `cargo install probe-run` which exits OK saying it's already installed. I then run `cargo run --release --bin dual-iir`, which says:
[image]

It seems to flash OK and get the network, but no MQTT server (as you'd expect - nothing has been set at this point). As the program hangs in here, I press `Ctrl+C` to terminate it, upon which it produces this log:
[image]

I then power off the device and then unconnect the probe. I connect the USB cable, then power the device back on at which point Windows gives me the warning 'USB device not recognised'. After this I try to run `python -m serial`, but the port doesn't show up.

Instead of this interface, I also tried to use the previous build method, where after `cd stabilizer` I run:
- `$env:BROKER='10.255.6.4'; cargo build --release`
- `$env:BROKER='10.255.6.4'; cargo objcopy --release --bin dual-iir -- -O binary dual-iir.bin`

I then plug the device in, power it on and run `$env:BROKER='10.255.6.4'; cargo run --release --bin dual-iir`. It flashes OK, gets on the network, but still no MQTT connection - as if the BROKER variable didn't have any effect on the build process.

However, if I hardcode the MQTT address in `src/hardware/flash.rs` (line 37), and I run the commands given in the manual at the moment (i.e. no env variable), it connects to MQTT alright:
[image]

This is the only way I can get it to work at the moment.

ryan-summers, 2023-12-01:
Hmm this is indeed suspicious - thanks for the detailed list of commands, it was helpful in ruling out some potential causes of this issue.

I suspect this is an issue with the usb-device stack, as I've already had to fix it a few times for various QUARTIQ projects.

juhaszp95, 2024-02-26:
Hello,

I've got the following output:
```
    Finished release [optimized + debuginfo] target(s) in 0.87s
     Running `probe-run --chip STM32H743ZITx --speed 30000 target\thumbv7em-none-eabihf\release\dual-iir`
⚠️  As of 11.10.2023 `probe-run` is in maintainance mode. We recommend everyone to switch to `probe-rs`. Read following article on the why and on how to migrate: https://ferrous-systems.com/blog/probe-run-deprecation/

────────────────────────────────────────────────────────────────────────────────
INFO - Starting
TRACE - Tracing
INFO - EUI48: 80-34-28-5f-59-0b
INFO - Found `/broker` in flash settings
INFO - Found `/id` in flash settings
INFO - setup() complete
WARN - Network link DOWN
TRACE - Suspending bus
TRACE - Control pipe reset
TRACE - EP0: setup=1, in_complete=0, out=0
TRACE - Read 8 bytes on EP0-OUT: [128, 6, 0, 1, 0, 0, 64, 0]
TRACE - EP0 request received: Request { direction: In, request_type: Standard, recipient: Device, request: 6, value: 256, index: 0, length: 64 }
TRACE - Processing Device::GetDescriptor
TRACE - wrote EP0: [18, 1, 16, 2, 2, 0, 0, 8]
TRACE - EP0: setup=0, in_complete=1, out=0
TRACE - wrote EP0: [9, 18, 47, 57, 16, 0, 1, 2]
TRACE - EP0: setup=0, in_complete=0, out=1
TRACE - Control transfer completed. Current state: DataIn
TRACE - EP0: setup=1, in_complete=0, out=0
TRACE - Read 8 bytes on EP0-OUT: [0, 5, 8, 0, 0, 0, 0, 0]
TRACE - EP0 request received: Request { direction: Out, request_type: Standard, recipient: Device, request: 5, value: 8, index: 0, length: 0 }
TRACE - Setting device address to 8
TRACE - Control pipe reset
TRACE - EP0: setup=1, in_complete=0, out=0
TRACE - Read 8 bytes on EP0-OUT: [128, 6, 0, 1, 0, 0, 64, 0]
TRACE - EP0 request received: Request { direction: In, request_type: Standard, recipient: Device, request: 6, value: 256, index: 0, length: 64 }
TRACE - Processing Device::GetDescriptor
TRACE - wrote EP0: [18, 1, 16, 2, 2, 0, 0, 8]
TRACE - EP0: setup=0, in_complete=1, out=1
TRACE - Control transfer completed. Current state: DataIn
TRACE - EP0: setup=1, in_complete=0, out=0
TRACE - Read 8 bytes on EP0-OUT: [0, 5, 9, 0, 0, 0, 0, 0]
TRACE - EP0 request received: Request { direction: Out, request_type: Standard, recipient: Device, request: 5, value: 9, index: 0, length: 0 }
TRACE - Setting device address to 9
TRACE - EP0: setup=0, in_complete=1, out=0
TRACE - EP0: setup=1, in_complete=0, out=0
TRACE - Read 8 bytes on EP0-OUT: [128, 6, 0, 1, 0, 0, 18, 0]
TRACE - EP0 request received: Request { direction: In, request_type: Standard, recipient: Device, request: 6, value: 256, index: 0, length: 18 }
TRACE - Processing Device::GetDescriptor
TRACE - wrote EP0: [18, 1, 16, 2, 2, 0, 0, 8]
TRACE - EP0: setup=0, in_complete=1, out=0
TRACE - wrote EP0: [9, 18, 47, 57, 16, 0, 1, 2]
TRACE - EP0: setup=0, in_complete=1, out=0
TRACE - wrote EP0: [3, 1]
TRACE - EP0: setup=0, in_complete=1, out=1
TRACE - Control transfer completed. Current state: DataInLast
TRACE - EP0: setup=1, in_complete=0, out=0
TRACE - Read 8 bytes on EP0-OUT: [128, 6, 0, 2, 0, 0, 255, 0]
TRACE - EP0 request received: Request { direction: In, request_type: Standard, recipient: Device, request: 6, value: 512, index: 0, length: 255 }
TRACE - Processing Device::GetDescriptor
TRACE - wrote EP0: [9, 2, 67, 0, 2, 1, 0, 128]
TRACE - EP0: setup=0, in_complete=1, out=0
TRACE - wrote EP0: [50, 9, 4, 0, 0, 1, 2, 2]
TRACE - EP0: setup=0, in_complete=1, out=0
TRACE - wrote EP0: [0, 0, 5, 36, 0, 16, 1, 4]
TRACE - EP0: setup=0, in_complete=1, out=0
TRACE - wrote EP0: [36, 2, 0, 5, 36, 6, 0, 1]
TRACE - EP0: setup=0, in_complete=1, out=0
TRACE - wrote EP0: [5, 36, 1, 0, 1, 7, 5, 129]
TRACE - EP0: setup=0, in_complete=1, out=0
TRACE - wrote EP0: [3, 8, 0, 255, 9, 4, 1, 0]
[... 188 lines cut ...]
TRACE - EP0: setup=0, in_complete=1, out=0
TRACE - wrote EP0: [105, 0, 110, 0, 97, 0, 114, 0]
TRACE - EP0: setup=0, in_complete=1, out=0
TRACE - wrote EP0: [97, 0]
TRACE - EP0: setup=0, in_complete=1, out=1
TRACE - Control transfer completed. Current state: DataInLast
TRACE - EP0: setup=1, in_complete=0, out=0
TRACE - Read 8 bytes on EP0-OUT: [128, 6, 2, 3, 9, 4, 255, 0]
TRACE - EP0 request received: Request { direction: In, request_type: Standard, recipient: Device, request: 6, value: 770, index: 1033, length: 255 }
TRACE - Processing Device::GetDescriptor
TRACE - wrote EP0: [22, 3, 83, 0, 116, 0, 97, 0]
TRACE - EP0: setup=0, in_complete=1, out=0
TRACE - wrote EP0: [98, 0, 105, 0, 108, 0, 105, 0]
TRACE - EP0: setup=0, in_complete=1, out=0
TRACE - wrote EP0: [122, 0, 101, 0, 114, 0]
TRACE - EP0: setup=0, in_complete=1, out=1
TRACE - Control transfer completed. Current state: DataInLast
WARN - Network link UP
INFO - MQTT connected, subscribing to settings
INFO - Opening stream
```

There's no more output, but the program doesn't terminate - so I press Ctrl+C to terminate.

The device now actually shows up, it's not saying it's a stabiliser, but it's listed as `---  3: COM4                 'USB Serial Device (COM4)'`. However, it's interesting to note that the re-flash does not erase the memory for the MQTT server that has been entered during an earlier flash (via Linux), so I'm not sure if this is somehow a remnant of that or something. (I.e. the device seems to remember what non-default MQTT settings it's been given before the reflash.)

ryan-summers, 2024-02-26:
It sounds like, based on the information you provided, that the USB issues have resolved in the branch I provided. What you describe is all the normal operational behavior of the device, so I'll work on updating the USB stack in the mainline stabilizer repository and get those fixes up-streamed.

juhaszp95, 2024-02-26:
I can confirm the main branch still doesn't show up on the list of USB devices on Windows. It does show up and say 'Stabilizer' on Linux. This new branch now shows up on Windows (but doesn't say Stabilizer, which would be useful).

ryan-summers, 2024-02-28:
> It does show up and say 'Stabilizer' on Linux. This new branch now shows up on Windows (but doesn't say Stabilizer, which would be useful).

Unfortunately this is a product of Windows that we can't control. Windows names the ports based on installed USB drivers, so we'd have to distribute a custom Windows USB driver to get Stabilizer to be named properly, which is rather complicated for simply identifying the correct serial port. As such, I don't think there's any activity that we can reasonably do to fix this.

closed 2024-02-28 by jordens

juhaszp95, 2024-04-09:
Sorry for having to reopen this issue, but I think the stabilizer is not detected on Windows again since the updates in the firmware.

ryan-summers, 2024-06-21:
@juhaszp95 Can you confirm that with the latest firmware, you're still seeing USB enumeration issues on Windows? If so, please reopen this issue so we can investigate.

juhaszp95, 2024-06-25:
I'm away at the moment, so cannot check, but perhaps @dnadlinger or @ayushagrawal2 can help.

PR #851 (2024-02-28): Fixing USB enumeration issues on Windows
Fixes #817 by patching to a fixed `usb-device` version. I'm planning on re-releasing `usb-device` as 0.3.2 shortly, so we can hold off until then if desired as well.