quartiq/stabilizer #822: support changing applications settings via serial-settings and load/store in flash

jordens, 2023-12-01:
* On boot the initial settings are loaded from flash, or `Default::default()` if invalid
* Then they are altered from retained, republished, and altered from non-retained pub messages on mqtt as currently done
* serial terminal allows accessing them (get and set) as well
* serial terminal allows comitting them to flash

Impact/feasibility to be determined.
This allows full standalone operation without network/broker.

ryan-summers, 2023-12-05:
Theoretically this is quite easy to implement. The main hurdle that I foresee is that the `Settings` structure is owned by the Miniconf client and the serial interface respectively, so there's some thought that would need to go into how to maintain synchronization between these two structures.

If we could somehow create a singleton of the settings that was referenced by the MQTT client and the serial interface, I believe this would become relatively trivial to implement.

jordens, 2023-12-05:
I would only ever modify stored settings over serial and only load settings from flash on boot (like broker and IP currently). That solves repub.

ryan-summers, 2023-12-05:
Ah so the intent here would be to make the serial interface effectively only touch flash, as opposed to modifying the run-time state of the device? It seems like that would be a bit... unexpected? I don't see anything blocking us from allowing the serial interface to affect the run-time state as well outside of software complexity.

jordens, 2023-12-05:
As for broker and IP. Those also need a reboot. If we call the flash settings to be the boot settings then all is well defined.
And as you say, repub races against potential host retained settings. Hard to clear all corner cases. Better leave simple settings precedence rules (default < flash < mqtt)

ryan-summers, 2023-12-05:
I agree w.r.t. the broker and IP settings being boot-time configuration, but the application run-time settings is what this issue is about, correct?

It seems like there would be limited usefulness to configuring run-time settings over serial if you had to save them and reboot the machine constantly to have them take effect.

I would expect that:
1. Runtime settings would immediately apply to the device operational state when configured via Serial
2. There would be an option for me to "save" the operational state via the serial console when I'm happy with the configuration
   * Maybe this would be automatic, similar to how we auto-save the broker and IP currently
3. I could harmoniously use the MQTT interface to configure as well, but these values would not be auto-saved to flash
4. There would be some MQTT command to save the device configuration to flash so they're used on the next boot

Am I missing something here?

jordens, 2023-12-05:
* There would quite some explaining, misunderstanding and potential frustration about why broker/ip need a reboot and e.g. streaming target doesn't.
* Imagine a retained setting in MQTT. Now you change that same setting over serial. Do you republish? Do you do so retained? Then you reboot the device. Loads the setting from flash, but the retained mqtt setting quickly overrides the flash setting. Confusion ensues (why did my setting via serial work just fine a while ago but again now doesn't).

ryan-summers, 2023-12-05:
Alright, the retained settings were indeed something I was concerned with as well. So the current thought is to expose a mechanism to configure + save runtime settings via the serial interface, but the serial interface never affects the current operating state of the device. I like this compromise.

Would we also want to expose a command over MQTT to save the current configuration?

jordens, 2023-12-05:
I wouldn't do that now. The use case where the boot settings (flash settings) are dominant (because there's no network connection) shall require USB connection.

PR #857 (2024-03-27): Adding runtime settings initial load from flash
This PR fixes #822 by adding support for modifying the runtime settings and persisting them to flash via the USB interface.

This should be merged after:
* #858
closed 2024-03-27 by jordens