quartiq/stabilizer #864: serial settings `list` should skip repeating default if equal

jordens, 2024-03-27:
`/dual_iir/iir_ch/0/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0}]`

In the case where they are default, just say

`/dual_iir/iir_ch/0/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]`

ryan-summers, 2024-03-28:
I took a look at this yesterday because I was thinking the same. The main issue with implementation is that we only have a single copy in memory at a time (i.e. either the default version, or the current value. We can't compare them because as soon as we're done with one,  we erase it to format the next.

ryan-summers, 2024-03-28:
Maybe some PartialEq implementation could help, but we don't know the concrete type of the settings in `serial-settings`, so we are operating with the JSON-formatted version

jordens, 2024-03-28:
It's a nice-to-have.
Maybe just split the serialize buffer for this?

jordens, 2024-03-29:
Or hash each after serialization.

PR #870 (2024-04-03): Omitting default value printing when values are default
Fixes #864 by checking if the current settings value and the default value are equivalent by comparing an FNV1A 64-bit hash of the serialized strings.

There's a small hash collision probability, but for our case here it is likely inconsequential because the user can always refer to source code.

```
/net/broker: "mqtt" [default]
/net/id: "04-91-62-d2-a8-6f" [default]
/net/ip: "0.0.0.0" [default]

> set /net/broker "test"
Settings saved. Reboot device (`platform reboot`) to apply.

> list
/dual_iir/afe/0: "G1" [default]
/dual_iir/afe/1: " [... cut]
closed 2024-04-03 by jordens