quartiq/stabilizer #885: usb serial terminal buffer issue

jordens, 2024-04-24:
I made the `iir_ch` array a bit longer (20x1) to show the issue in `dual-iir`:

When hitting `list` on a larger struct over the serial terminal, it truncates output at some variable length and doesn't print everything. I played with the buffers a bit but couldn't quickly figure out where the issue was.

```
> listii
/dual_iir/afe/0: "G1" [default]
/dual_iir/afe/1: "G1" [default]
/dual_iir/iir_ch/0/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/1/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/2/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/3/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/4/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/5/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/6/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/7/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/8/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/9/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/10/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/11/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/12/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/13/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/14/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/15/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/16/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/17/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/18/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/19/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/allow_hold: false [default]
/dual_iir/force_hold: false [default]
/dual_iir/telemetry_period: 10 [default]
/dual_iir/stream_target: {"ip":[0,0,0,0],"port":0} [default]
/dual_iir/signal_generator/0/signal: "Cosine" [default]
/dual_iir/signal_generator/0/frequency: 1000.0 [default]
/dual_iir/signal_generator/0/symmetry: 0.5 [default]
/dual_iir/signal_generator/0/amplitude: 0.0 [default]
/dual_iir/signal_generator/0/phase: 0.0 [default]
> listiir/si
/dual_iir/afe/0: "G1" [default]
/dual_iir/afe/1: "G1" [default]
/dual_iir/iir_ch/0/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/1/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/2/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/3/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/4/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/5/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/6/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/7/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/8/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/9/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/10/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/11/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/12/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/13/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/14/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/15/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/16/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/17/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":32767.0} [default]
/dual_iir/iir_ch/18/0: {"ba":[1.0,0.0,0.0,0.0,0.0],"u":0.0,"min":-32767.0,"max":327
```

ryan-summers, 2024-06-20:
So ultimately this is governed by the underlying `SerialPort` buffer size, which is 1024 bytes. The reason we can send more than that is because on each USB serial write, we can potentially send 64 bytes out the USB interface (if the EP is in a state to transmit).

I supect that because the serialized structure of the IIR channel is long, this can cause the USB buffer to back up faster than it can send data out. If we wanted to guarantee that this would be resolved, we would need to increase the `SerialPort` write_store buffer size.

I suspect that Booster does not have this issue (even though it can print a large settings structure) because the CPU is running at a slower speed than Stabilizer, which means it takes longer actual amounts of time to progress through listing, serializing, and transmitting the settings structure, which allows more data to be transmitted via USB while the CPU is processing data.

jordens, 2024-06-20:
Isn't there a way to keep the buffer small and make this async "blocking", i.e. use an `async fn write()` that can suspend for another task to drain the buffer?