quartiq/stabilizer #903: Streaming off by a factor of 2

juhaszp95, 2024-05-30:
Hello,

It seems to me that the streaming of voltages from the dual-iir application is off by a factor of 2. For example, if I put a 1 Vpp signal on the input of the stabilizer, it reports a 2 Vpp signal.

The signal generator I am using expects a 50 ohm load, but presumably the stabilizer is 50 ohm - in which case it should receive a 1 Vpp signal. (It would receive a 2 Vpp signal if it was a high-impedance input.) I've also checked the output of my signal generator on an oscilloscope, just be sure, and it is correct.

jordens, 2024-05-30:
Please supply some data! What does "it reports" mean?
Don't presume, instead please check and set the 50 R termination on stabilizer the way you need it and so that it matches your scope input.

juhaszp95, 2024-05-30:
What do you mean by `check and set the 50 R termination on stabilizer`? Is the input impedance of the stabilizer not 50 ohm?

I've actually just measured this with a multimeter, and it seems that input impedance is 200 kohm, while the output impedance is 50 ohm - in which case there might not be an issue with the streaming after all.

Could you confirm this is indeed the case, or where I could find the design specifications? Is there any way to set a different input impedance on the board?

jordens, 2024-05-30:
>  Is the input impedance of the stabilizer not 50 ohm?

It is what you set it. Have a look!

closed 2024-05-30 by jordens

juhaszp95, 2024-05-30:
I'm sorry, but I haven't found any documentation saying where it can be said. Could you point me to the specific location?

jordens, 2024-05-30:
The board, the front panel!

juhaszp95, 2024-05-30:
Is there any documentation regarding how to do this? I can't see any switches or similar on the front panel. Any instructions or documentation would be appreciated.

jordens, 2024-05-31:
Best place to get help on that front should be where you got the device or the hardware repo. This repo here is about a specific firmware.

dnadlinger, 2024-05-31:
For posterity:
[image]
(or similar on newer revisions, though this really isn't the best place to ask hardware questions)