quartiq/stabilizer #910: ITCM init

jordens, 2024-06-19:
https://github.com/quartiq/stabilizer/blob/3ef00cbd92b9ee57b19026ba7abb3791e88be339/src/hardware/setup.rs#L183

That destination slice is uninitialized, thus insta-UB. Needs to be `MaybeUninit`.

ryan-summers, 2024-06-28:
Not only that, but the pointer provenance of the `sitcm` pointer only permits us to technically write 4 bytes (because the `static mut __sitcm: u32`.

ryan-summers, 2024-06-28:
Ref https://github.com/rust-embedded/wg/issues/771, it might be the case that pointer provenance is not actually an issue

jordens, 2024-06-28:
Way back I had [argued](https://github.com/quartiq/stabilizer/pull/322#discussion_r632622707) that this is like an FFI interface where once crates slices from addresses and lengths in a similar way without virtual machine knowing much about them.
I.e. specifically the `src` slice must be fine to create like this. How else would one do it?
And if the `src` slice is fine (initialized, doesn't break aliasing rules), so is the `dst` slice.

jordens, 2024-07-09:
Closing as "not a problem".

closed 2024-07-09 by jordens