quartiq/stabilizer #931: [v1.3.2 hw] DAC8831IBD noise

MorganTL, 2024-07-22:
Flashed ff2c2c1 on v1.3.2 hardware, both DAC8831IBD outputs are very noisy.

## Test setup
### With ADC input
- Connect 1Vpp 10kHz sine wave to ADC and oscilloscope blue channel using T splitter
- Connect DAC to oscilloscope yellow channel

[image]

### Using `signal_generator`
- set amplitude to 1, signal to Cosine, frequency to 1000, symmetry to 0.5, phase to 0.0 using usb serial
- Connect DAC to oscilloscope yellow channel

[image]

## Related issue
- [sinara-hw/Stabilizer #119](https://github.com/sinara-hw/Stabilizer/issues/119)

jordens, 2024-07-22:
Certainly doesn't look like a DAC issue and potentially not even a software issue then. Could you please check the ADC data?

jordens, 2024-07-22:
And show me `platform service`

jordens, 2024-07-22:
And the manufacturer will aid you with hardware issues.

ryan-summers, 2024-07-22:
The linked issue implies this was built by Creotech, and the further information indicates this is indeed a DAC issue, as replacement of the DAC8831 with the AD5542A was claimed to have resolved this issue.

It seems like there is a concern that SPI timing characteristics may not be being met with the new DAC, but I don't see a significant difference between the two while taking a cursory glance.

jordens, 2024-07-22:
@ryan-summers I'd be more cautious about those conclusions. I don't see how the issue implies manufacturer. The fact that excluding the ADC also makes the issue go away points elsewhere. Please let @MorganTL respond to my requests.

MorganTL, 2024-07-23:
> And the manufacturer will aid you with hardware issues. Who is it?

The card is a production sample from us (M-labs). And I am the electronics engineer who debugged the hardware issue. I raised an issue here because @gkasprow suggested [it maybe an SPI issue](https://github.com/sinara-hw/Stabilizer/issues/119#issuecomment-2236306422).

> Certainly doesn't look like a DAC issue and potentially not even a software issue then. Could you please check the ADC data?
> And show me `platform service`

Unfortunately, we returned the production sample to our PCBA provider for rework.

jordens, 2024-07-23:
Non-reproducible then. Feel free to reopen when you get it back, can reproduce it, and have the requested info.

closed 2024-07-23 by jordens