quartiq/stabilizer #1020: Miniconf Unicode and Basic Syntax

pyquest1, 2025-05-09:
I am trying to get our stabilizer up and running and could use some help.  I went through the documentation and found an issue with Miniconf - came up with a possible solution and am unsure if I am communicating correctly.

- Computer and System  - Ubuntu - python 3.12.8 - trying to use the nix package manager.

1.) Status
The stabilizer card has been successfully flashed with the lockin.bin file.

2.) I was able to install miniconf 

`python -m pip install miniconf`

but then upon running 

`python -m miniconf --help` i received compiler, 

`python -m pip install miniconf
Collecting miniconf
  Using cached miniconf-1.2.0.tar.gz (20 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: miniconf
  Building wheel for miniconf (pyproject.toml) ... done
  Created wheel for miniconf: filename=miniconf-1.2.0-py3-none-any.whl size=7474 sha256=463cb00dbf808d3fa88f5deaac61da231d3f248dfe19e775ae64653287d141ce
  Stored in directory: /home/grpartiq/.cache/pip/wheels/70/6a/a1/7f6433bc57362c2f0ddb2b3d47ae5e6a8125b5325e88c22f4a
Successfully built miniconf
Installing collected packages: miniconf
Successfully installed miniconf-1.2.0

[notice] A new release of pip is available: 24.3.1 -> 25.1.1
[notice] To update, run: pip install --upgrade pip
(vpy) grpartiq@hrlit3085:~/ARTIQ/Nix_Tests$ python -m miniconf --help
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/grpartiq/ARTIQ/Nix_Tests/vpy/lib/python3.12/site-packages/miniconf.py", line 114, in <module>
    import compiler
ModuleNotFoundError: No module named 'compiler'
(vpy) grpartiq@hrlit3085:~/ARTIQ/Nix_Tests$ python -m miniconf --help
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/grpartiq/ARTIQ/Nix_Tests/vpy/lib/python3.12/site-packages/miniconf.py", line 114, in <module>
    import compiler
ModuleNotFoundError: No module named 'compiler'
(vpy) grpartiq@hrlit3085:~/ARTIQ/Nix_Tests$ python -m pip install compiler
Collecting compiler
  Using cached compiler-0.2.0-py3-none-any.whl.metadata (2.1 kB)
Using cached compiler-0.2.0-py3-none-any.whl (5.6 kB)
Installing collected packages: compiler
Successfully installed compiler-0.2.0

[notice] A new release of pip is available: 24.3.1 -> 25.1.1
[notice] To update, run: pip install --upgrade pip
(vpy) grpartiq@hrlit3085:~/ARTIQ/Nix_Tests$ python -m miniconf --help
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/grpartiq/ARTIQ/Nix_Tests/vpy/lib/python3.12/site-packages/miniconf.py", line 204, in <module>
    class _Dump:
  File "/home/grpartiq/ARTIQ/Nix_Tests/vpy/lib/python3.12/site-packages/miniconf.py", line 205, in _Dump
    types = (dict, list, tuple, int, float, str, unicode, bool, type(None))
                                                 ^^^^^^^
NameError: name 'unicode' is not defined
grpartiq@hrlit3085:~/ARTIQ/Nix_Tests$ python -m miniconf --help
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/grpartiq/ARTIQ/Nix_Tests/vpy/lib/python3.12/site-packages/miniconf.py", line 204, in <module>
    class _Dump:
  File "/home/grpartiq/ARTIQ/Nix_Tests/vpy/lib/python3.12/site-packages/miniconf.py", line 205, in _Dump
    types = (dict, list, tuple, int, float, str, unicode, bool, type(None))
                                                 ^^^^^^^
NameError: name 'unicode' is not defined
(vpy) grpartiq@hrlit3085:~/ARTIQ/Nix_Tests$ python -m pip install unicode
Collecting unicode
  Downloading unicode-2.9-py2.py3-none-any.whl.metadata (1.1 kB)
Downloading unicode-2.9-py2.py3-none-any.whl (14 kB)
Installing collected packages: unicode
Successfully installed unicode-2.9

grpartiq@hrlit3085:~/ARTIQ/Nix_Tests$ python -m miniconf --help
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/grpartiq/ARTIQ/Nix_Tests/vpy/lib/python3.12/site-packages/miniconf.py", line 204, in <module>
    class _Dump:
  File "/home/grpartiq/ARTIQ/Nix_Tests/vpy/lib/python3.12/site-packages/miniconf.py", line 205, in _Dump
    types = (dict, list, tuple, int, float, str, unicode, bool, type(None))
                                                 ^^^^^^^
NameError: name 'unicode' is not defined

`

So then I gave up and modified miniconf.py lines 204-219 to replace unicode with bytes.  Following: https://stackoverflow.com/questions/19877306/nameerror-global-name-unicode-is-not-defined-in-python-3

`
class _Dump:
    types = (dict, list, tuple, int, float, str, bytes, bool, type(None))
        
    def __init__(self):
        self.hexpr = re.compile('^[^#]', re.MULTILINE)
            
    def __call__(self, data, headers, pedantic, **kw):
        data = self.recast(data, pedantic, [dict])
        return '\n'.join(
            [self.header(headers, '--top--', **kw)] +
            ['%s%s = %s\n' %
             (self.header(headers, k, **kw),
              k, pprint.pformat(data[k], **kw))
             for k in self.keys(data)] +
            [self.header(headers, '--bottom--', **kw)]
            ).strip()
        `

This caused the package to install

`(vpy) grpartiq@hrlit3085:~/ARTIQ/Nix_Tests$ python -m miniconf --help
(vpy) grpartiq@hrlit3085:~/ARTIQ/Nix_Tests$ python -m miniconf --broker 10.5.44.55 dt/sinara/dual-iir/00-11-22-33-44-55 stream='"10.34.16.123:4000"'
(vpy) grpartiq@hrlit3085:~/ARTIQ/Nix_Tests$ python -m miniconf --broker enum Signal {Cosine,}
(vpy) grpartiq@hrlit3085:~/ARTIQ/Nix_Tests$ python -m miniconf --broker enum Source  {1,0,250000}
`

but I am not seeing any frequencies/amplitudes come out of the DAC's with the miniconf commands.  Can someone help me verify the install is working correctly?

jordens, 2025-05-09:
That miniconf has nothing to do with stabilizer. Please follow the documentation.

closed 2025-05-09 by jordens

pyquest1, 2025-05-09:
Hi Jordens,

I am following the guide explicitly https://quartiq.de/stabilizer/usage.html .  Can you help me understand how to communicate with the device and validate the documented way of communicating with the unit if not with miniconf?

pyquest1, 2025-05-09:
> That miniconf has nothing to do with stabilizer. Please follow the documentation.

Hi Jordens,

I am following the guide explicitly https://quartiq.de/stabilizer/usage.html . Can you help me understand how to communicate with the device and validate the documented way of communicating with the unit if not with miniconf?

jordens, 2025-05-10:
No you aren't. Already your first step in 2 ignores the docs.

pyquest1, 2025-05-12:
> No you aren't. Already your first step in 2 ignores the docs.

@jordens 

Jodens,

I promise we did not ignore it.  I just didn't post the entire log.  The issue is that miniconf is not installing per the documentation.

`python -m venv --system-site-packages vpy

# Refer to https://docs.python.org/3/tutorial/venv.html for more information on activating the
# virtual environment. This command is different on different platforms.
./vpy/Scripts/activate

Next, install prerequisite packages

python -m pip install py

To use miniconf, execute it as follows:

python -m miniconf --help

Miniconf also exposes a programmatic Python API, so it's possible to write automation scripting of Stabilizer as well.`

To which I originally tried:

`[nix-shell:~/ARTIQ/stabilizer]$ python -m venv --system-site-packages vpy

[nix-shell:~/ARTIQ/stabilizer]$ ./vpy/Scripts/activate
bash: ./vpy/Scripts/activate: No such file or directory

[nix-shell:~/ARTIQ/stabilizer]$ python -m pip install py
/nix/store/y5d7i2n2xcxgprxjn3bbkhiwprmi9s6f-python3-3.12.8-env/bin/python3.12: No module named pip

[nix-shell:~/ARTIQ/stabilizer]$ python -m miniconf --help
/nix/store/y5d7i2n2xcxgprxjn3bbkhiwprmi9s6f-python3-3.12.8-env/bin/python3.12: No module named miniconf
`

jordens, 2025-05-13:
You need to properly format your posts and provide complete information.
Your pip is apparently newer and has different precedence. Try `python -m pip install py/`.

pyquest1, 2025-05-17:
> You need to properly format your posts and provide complete information. Your pip is apparently newer and has different precedence. Try `python -m pip install py/`.

That didn't work either

`
grpartiq@hrlit3085:~ /ARTIQ/Artiq_Workplace$ python -m pip install py/
/nix/store/y5d7i2n2xcxgprxjn3bbkhiwprmi9s6f-python3-3.12.8-env/bin/python3.12: No module named pip
grpartiq@hrlit3085:~ /ARTIQ/Artiq_Workplace$ pip

Usage:   
  pip <command> [options]

Commands:
  install                     Install packages.
  download                    Download packages.
  uninstall                   Uninstall packages.
  freeze                      Output installed packages in requirements format.
  list                        List installed packages.
  show                        Show information about installed packages.
  check                       Verify installed packages have compatible dependencies.
  config                      Manage local and global configuration.
  search                      Search PyPI for packages.
  cache                       Inspect and manage pip's wheel cache.
  index                       Inspect information available from package indexes.
  wheel                       Build wheels from your requirements.
  hash                        Compute hashes of package archives.
  completion                  A helper command used for command completion.
  debug                       Show information useful for debugging.
  help                        Show help for commands.

General Options:
  -h, --help                  Show help.
  --debug                     Let unhandled exceptions propagate outside the main subroutine, instead of logging them to stderr.
  --isolated                  Run pip in an isolated mode, ignoring environment variables and user configuration.
  --require-virtualenv        Allow pip to only run in a virtual environment; exit with an error otherwise.
  -v, --verbose               Give more output. Option is additive, and can be used up to 3 times.
  -V, --version               Show version and exit.
  -q, --quiet                 Give less output. Option is additive, and can be used up to 3 times (corresponding to WARNING, ERROR, and CRITICAL
                              logging levels).
  --log <path>                Path to a verbose appending log.
  --no-input                  Disable prompting for input.
  --proxy <proxy>             Specify a proxy in the form [user:passwd@]proxy.server:port.
  --retries <retries>         Maximum number of retries each connection should attempt (default 5 times).
  --timeout <sec>             Set the socket timeout (default 15 seconds).
  --exists-action <action>    Default action when a path already exists: (s)witch, (i)gnore, (w)ipe, (b)ackup, (a)bort.
  --trusted-host <hostname>   Mark this host or host:port pair as trusted, even though it does not have valid or any HTTPS.
  --cert <path>               Path to PEM-encoded CA certificate bundle. If provided, overrides the default. See 'SSL Certificate Verification'
                              in pip documentation for more information.
  --client-cert <path>        Path to SSL client certificate, a single file containing the private key and the certificate in PEM format.
  --cache-dir <dir>           Store the cache data in <dir>.
  --no-cache-dir              Disable the cache.
  --disable-pip-version-check
                              Don't periodically check PyPI to determine whether a new version of pip is available for download. Implied with
                              --no-index.
  --no-color                  Suppress colored output.
  --no-python-version-warning
                              Silence deprecation warnings for upcoming unsupported Pythons.
  --use-feature <feature>     Enable new functionality, that may be backward incompatible.
  --use-deprecated <feature>  Enable deprecated functionality, that will be removed in the future.
grpartiq@hrlit3085:~/ARTIQ/Artiq_Workplacgrpartiq@hrlit3085:~/ARTIQ/Artiq_Workplacgrpartiq@hrlit3085:~/ARgrpartiq@hrlit3085:~/ARTIQ/Artiq_Workplace$grpartiq@hrlit3085:~/ARTIQ/Artiq_Workgrpartiq@hrlit3085:~/ARgrpartiq@hrlit3085grpartiq@hgrpartiqgrpartiq@hrlgrpagrpartgrpagrpagrpartiqgrpartgrgrgrgrpartiq@hrlit3085:~/ARTIQ/Artiq_Workplace$ python -m pip install py
/nix/store/y5d7i2n2xcxgprxjn3bbkhiwprmi9s6f-python3-3.12.8-env/bin/python3.12: No module named pip

`

however I did get it to install via flakes

`

  inputs.miniconf.url = "github:quartiq/miniconf";
  miniconf.packages.x86_64-linux.miniconf-mqtt-py
[... 4 lines cut ...]

`
I have no name!@hrlit3085:~/ARTIQ/Artiq_Workplace$ python -m miniconf -b mqtt -d dt/sinara/lockin/+ stream='"10.5.44.55:1883"'
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/nix/store/fc2r67lldh3j5mffp3qb5cbdqg9wxqps-python3.12-miniconf-0.18.0/lib/python3.12/site-packages/miniconf/__main__.py", line 7, in <module>
    asyncio.run(_main())
  File "/nix/store/f2krmq3iv5nibcvn4rw7nrnrciqprdkh-python3-3.12.9/lib/python3.12/asyncio/runners.py", line 195, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/nix/store/f2krmq3iv5nibcvn4rw7nrnrciqprdkh-python3-3.12.9/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
  File "/nix/store/fc2r67lldh3j5mffp3qb5cbdqg9wxqps-python3.12-miniconf-0.18.0/lib/python3.12/site-packages/miniconf/async_.py", line 272, in _main
    async with Client(args.broker, protocol=MQTTv5) as client:
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/gaai15q1x3f3kx13qmfabkyhp9ji7kyk-python3.12-aiomqtt-2.0.1/lib/python3.12/site-packages/aiomqtt/client.py", line 723, in __aenter__
    raise MqttError(str(exc)) from None
aiomqtt.exceptions.MqttError: [Errno -3] Temporary failure in name resolution

`